fietser28/EL-Load #1: Optimise MOSFET placement on heat sink and/or power distribution between MOSFETS.

fietser28, 2024-01-22:
We need to find a good placement of the 4 MOSFETS on the LAM 3 150 heat sink. The MOSFETs are fixed on the PCB and it would require a special setup to move them around or a lot of different PCB’s to test different configurations.

A form of thermal analysis simulation to find an optimal/good placement seems more feasible. The analysis is not very trivial because the fan cases a forced airflow thru the heatsink and the air will heat up while flowing thru the heatsink.

On my current test PCB the MOSFETS are evenly placed along the heatsink and all MOSFETS dissipate the same amount of power (this is forced by the circuit). This situation results in the MOSFETS further away from the fan being around 15-20 degrees Celsius higher. The maximum temperature of the individual MOSFETs places the limit on the maximum power rating of the device. In my test I can now dissipate 90W total in a 19 degrees room temperature. The heatsink close to the MOSFET with highest temperature is just below 90 degrees Celsius. 

The MOSFETS are IRF250 in TO-247 package with thermal grease and a kapton foil isolation. They are pressed against the heatsink with DIY clamps at the place of the screw hole of the package. Because of the experimentation I don’t want to drill a lot of holes in the heatsink.

Thermal images show a gradual increase in heatsink temperature when moving away from the fan. The increase is comparable to the temperature difference between the MOSFETS. Therefore I think it is worth investigating if placing all the MOSFETS closer to the fan results in a possible lower maximum MOSFET temperature at the same overall power dissipation.

A good/optimal placement maximises the total power the mosfets can dissipate without individually overheating.
For this I think there are 2 degreees of freedom we should focus on:
1.  placement of the 2 MOSFETS with the greatest distance from the fan. The other mosfets should be placed close to the fan.
2. Create a Fixed difference (in %) between currents (which is directly proportional to the power) of the MOSFETS close and far away of the fan. If simulation shows this is beneficial it is straight forward to create this fixed difference in the circuit.

I think we should first optimise on 1 and second on 2.

jbb-nz, 2024-02-17:
I expect that a big factor in the temperature difference is due to the air inside the heatsink warming up, so I’m not sure how much difference item 1 (fiddling with the ‘downwind’ pair of MOSFETs’ position) will make.

I do think that item 2 could be a win.

fietser28, 2024-03-03:
I also have a few other factor to play with:

3. Thermal conductivity from MOSFETs to heatsink. Use grease, karnton, other type of pads.
4. Fan. I have several fans. 12V fans seem more powerful. Will also test with this.

For more structural testing I have an studio dashboard that increases the power, waits until the temperature measurement on the heat sink stabilises and records this before taking the next step.

only the physical placement is hard to change , to others are probably doable.

fietser28, 2024-03-10:
I've tested item 2: change the current ratio between MOSFETS. This doesn't change the overall heatsink temperature but it does change the individual MOSFETS temperature (which is the limiting factor).

It turns out with the current 55mm distance between the MOSFETS a 1:2 ratio results in almost leveling the temperature between the MOSFET cases. And resulting in a lower max individual MOSFET temperature around 15C at the same power level.

For 90W this means 30W for a MOSFET near the fan and 15W in the middle.
I did not expect the ratio to be so big. Definitely something to keep in there. I implemented this using simple resistor dividers at the current setting.

fietser28, 2024-03-10:
item 4: I've tested with a more powerful 12V fan: This definitely results in more cooling. Except for the cost of small powerful fans this is the way to go.

jbb-nz, 2024-03-11:
Wow, a 30 W : 15 W ratio on the current sharing is really surprising. I wonder if the ambient temperature makes a big impact here? You previously mentioned 19 degrees C; where would the temperatures end up at an ambient of 35 degrees C (for a box jammed into a corner)?

If you are implementing circuitry to manage how the current reference is send to each opamp + MOSFET, maybe you could also implement a ‘low current’ range which only sends it to the two 15 W channels? I’ve found it very useful in the past to have a ‘low’ and ‘high’ range on an electronic load.

fietser28, 2024-03-11:
I was also surprised about the ratio. I was expecting more like 20%, but it makes sense.

I expect with the ambient temperature is linear with the heatsink temperature: at a constant ambient temperature I stepped the power up and waited for the temp to stabilize: the temperature rose linear with the power.

regarding the low ranges: What is your use case / preferred range?
I have a low range implemented for testing now: it only changes the measurement from 10A full scale to 1A full scale. This increases the accuracy, hopefully from 10mV accuracy and 1mA resolution to 1mA accuracy and. 0.1 mA resolution . Setting current stays the same: 1mA.
Voltage goes from 80V full scale to around 10V full scale.

jbb-nz, 2024-03-12:
Oops, I forgot about existing low range setting. 0.1 mA readout resolution is good. 1 mA setting resolution is probably just fine.