fietser28/EL-Load #11: Test with powerful fan

fietser28, 2024-03-10:
Test with powerfull fan: [MR3010E with 12V with PWM](1570-MR3010E12B1+6-RSR-ND)  ([datasheet](https://www.mechatronics.com/pdf/MR3010.pdf)).

Disadvantage: expensive and only available from Digikey. It really has to make a difference.

fietser28, 2024-05-03:
No noticable difference.

closed 2024-05-03 by fietser28

prasimix, 2026-10-23:
More affordable models could be found on TME.eu, e.g. 30x30x15mm [MF30151V1-1000U-G99](https://www.tme.eu/hr/en/details/mf30151v1-g99-a/dc12v-fans/sunon/mf30151v1-1000u-g99/) or 30x30x10mm [MF30101V1-1000U-G99](https://www.tme.eu/hr/en/details/mf30101v1-g99-a/dc12v-fans/sunon/mf30101v1-1000u-g99/). OK, these aren't PWM models, but I think they can serve the purpose quite well; they can simply operate as ON or OFF, depending on whether the dissipation has exceeded a set threshold. They are also available at Mouser.